kamilsss655/ESPRI #4: [FEATURE] Find a way to power the ESP from Kenwood connector

kamilsss655, 2024-03-04:
### Is your feature request related to a problem? Please describe.
To keep the compact form factor it would be great to power the ESP from the host radio through the Kenwood connector.

@spm81 tested and found out 3.3V on the Kenwood connector from K5, however it is not enough to power ESP from it (as it is likely output through 10K resistor like Baofeng).
 
### Describe the solution you'd like
Find out best way to get power through Kenwood connector to power the ESP.
This includes simple modifications to the host radio. In fact this has been done before with a Baofeng as described here:
https://slomkowski.eu/tutorials/baofeng-uv-5r-constant-power-connector-hack/

### Additional context
Kenwood connector pinout for K5:
https://github.com/ludwich66/Quansheng_UV-K5_Wiki/wiki/Programming-Cable

dobrishinov, 2024-03-11:
Well done on the firmware and smart move using the ESP32 with messaging features on the Meshtastic protocol! This radio will be super helpful in emergencies. I had a similar idea before, about making a messaging device with an ESP32 and packet protocol for Baofeng radios. I'm just wondering if it's doable to get power straight from the Type-C port of the UV-K5 instead of the Kenwood port. It would be handy if the port could switch some voltage around to make it easier to build a custom PCB with all the ports needed (Kenwood and Type-C). Adding an ESP32 onto this PCB would make everything easy to plug in and use. Or, we could consider getting power straight from the charging pins of the battery on the back, and maybe make a 3D printed docking station for that.

KDMcMullan, 2024-03-27:
I've got [a similar project](https://github.com/KDMcMullan/HaMQTT), using the ESP as an interface between the radio an MQTT broker. I was also hoping to power it from the Kenwood connector.
After quite an extensive search, it seems that we can buy a K-connector which has:

- DATA only (2.5mm ring, 2.5mm sleeve, 3.5mm sleeve) connected OR
- AUDIO only connector (2.5mm tip, 2.5mm sleeve, 3.5mm rig, 3.5mm sleeve).

As you know, power out is on 3.5mm tip. SOME data cables support this line, but I've not found an audio cable which does, and I've never seen a K-cable which has all 6 conductors connected. I find this surprising.

[This author](https://github.com/skuep/AIOC) solves the problem by building his own K connector onto a PCB from two independent plugs.

kamilsss655, 2024-03-31:
So it seems that ESP32 cannot be powered by stock K5 Kenwood output. The voltage drops from 3.3V to 1.7V as soon as ESP32 is connected. This leaves at least two options:
1) Powering ESP32 externally or via battery (lolin lite board supports 1S lipo)
2) Modifying the K5 to provide 8.4V straight from the radio's battery via Kenwood connection (as documented here: http://www.hamskey.com/2024/03/c-board-for-uv-k5.html), this would require ESPRI PCB to include a LDO

It might be different with different radios, and ESPRI is supposed to be host agnostic so hard assumptions about the host shouldn't be made.

kamilsss655, 2024-04-04:
The issue with K5 power might be resolved by adding 220-470uF capacitor to ESP. During boot it requires ~500mA, and the K5 boards do not have big capacitors on 3.3V rail so this might cause boot issues.

KDMcMullan, 2024-04-05:
That is agitating. My observation on the Quansheng UV-5R, is that the power available on the 3.5mm tip is an INPUT to the radio, which charges it. Maybe it's not supposed to be an outlet?

kamilsss655, 2024-04-11:
So adding 400uF capacitor to stock K5 didn't allow the ESP32 to boot up. Also the capacitors charged very slowly.

Perhaps bypassing the 10 ohm resistor while using the capacitor might help.

KDMcMullan, 2024-07-06:
OKay. Here's another thought: The board and the radio have to be powered somehow. Certainly, it would be nice to be able to use the radio's battery, but another option would be to use the board's supply to keep the radio battery topped up. I recognise that these HTs never fully charge from that port, but it does mean that having an external LiPo or whatever, isn't exactly "wasteful".

KDMcMullan, 2024-07-06:
Other thoughts: The HT battery charging ports provide the battery voltage. I don't know if they provide the current, though... Perhaps what's required is a 3D printed clip for the base of the HT, which contains a couple of pogo pins to power the board?

[image]

KDMcMullan, 2024-07-06:
> Yea, the ESPRI runs from K5 battery directly, so the docking station approach is one of the possible form factors. In fact one could use the included K5 charging station and place the PCB board inside it.

I'm not sure I understand. I thought it wasn't possible to power the ESPRI from the K5 battery? (At least, not via the K plug.) That's why I suggested this option. Have you already connected it directly to the battery by the rear conductors?

kamilsss655, 2024-07-06:
> Have you already connected it directly to the battery by the rear conductors?

Yea I connected K5 battery rear conductors to ESPRI `EXT 5-16V` header pins and it works correctly.

I haven't modified the K5 to output battery voltage yet, but it should also work. That's the reason for the 3 pad solder bridge on the ESPRI board so user can choose power path from the radio to either be 3.3V or 5-16V, by shorting the `RV+` pad to either `3.3V` or `16V` pad.