kamilsss655/ESPRI #5: [FEATURE] Sample audio input and encode it as PDM (ADC)

kamilsss655, 2024-03-04:
### Is your feature request related to a problem? Please describe.
We need to be able to sample audio inputs as part of building block for digital modes and scrambler.

### Describe the solution you'd like
Sample 2 seconds of audio output from radio (via Kenwood connector) using analog comparator of the ESP32 (or other means) and save it as PDM in memory.

### Additional context
https://en.wikipedia.org/wiki/Pulse-density_modulation
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdm.html
https://docs.espressif.com/projects/esp-idf/en/stable/esp32h2/api-reference/peripherals/ana_cmpr.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

KDMcMullan, 2024-03-27:
I see a ton of things you could do with it ... what's you plan for the feature? Do you know if the ESP can sample at a good audio rate?

kamilsss655, 2024-03-27:
So far I was able to sample audio input at 20kHz, so I assume it could work well. Currently I am a bit stuck because of the internals of ESP-IDF libraries and the resource management (it seems one can only transmit audio or sample audio at a given time).

KDMcMullan, 2024-03-27:
Would you want to do both simultaneously? Or are you saying that's two separate software builds needed?

kamilsss655, 2024-03-27:
I don't need to do both at the same time. Although I am having issue switching between rx and tx of audio. It might be I'm doing something wrong, but just in case I opened issue at: https://github.com/espressif/esp-idf/issues/13484

kamilsss655, 2024-03-28:
resolved in v.0.3.8

closed 2024-03-28 by kamilsss655