kamilsss655/ESPRI #7: [FEATURE] Create WIKI

kamilsss655, 2024-03-10:
### Is your feature request related to a problem? Please describe.
To bring more users to the project a proper WIKI page should be established.

### Describe the solution you'd like
Create a [WIKI](https://github.com/kamilsss655/ESPRI/wiki) section in the part of this repository.

### Describe alternatives you've considered
We could probably have a Sphinx page for a bit prettier documentation at some point.
It could look like this: https://sphinx-rtd-theme.readthedocs.io/en/stable/
That could be auto-deployed and hosted on Github.

ErikS-web, 2024-04-11:
I think ...

[image]

for this project called ...
[image]

Come 2 [Online Manual / Wiki](https://github.com/kamilsss655/ESPRI/wiki)

ErikS-web, 2024-04-11:
If you have an 

* idea for the Wiki, [create an Issue with label **[Documentation]**](https://github.com/kamilsss655/ESPRI/issues/new)

kamilsss655, 2024-04-11:
Great! Wiki looks awesome and will be of great help!

Thanks Erik!