kamilsss655/ESPRI #9: [BUG] release v.0.2.5 boot loop on TTGO T3_V1.6?

greghall-au, 2024-03-19:
Hi,
I'd really like to see this working, but don't (yet) have the ESP32 devices you specify, only TTGO T3_V1.6, so i try anyway.
But bootloop. What am I doing wrong? Maybe just wait for ESP32 Lolin Lite to arrive?
-
esptool.py erase_flash  OK
esptool.py write_flash 0x0 espri_v.0.2.5.bin
...
Flash will be erased from 0x00000000 to 0x001fffff...
Compressed 2097152 bytes to 687160...
Wrote 2097152 bytes (687160 compressed) at 0x00000000 in 62.0 seconds (effective 270.4 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...

-from serial port-
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15540
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3904
entry 0x40080640
.[0;32mI (63) boot: ESP-IDF v5.2.1 2nd stage bootloader.[0m
.[0;32mI (63) boot: compile time Mar 19 2024 11:00:20.[0m
.[0;32mI (63) boot: Multicore bootloader.[0m
.[0;32mI (68) boot: chip revision: v1.0.[0m
.[0;32mI (71) boot.esp32: SPI Speed      : 40MHz.[0m
.[0;32mI (76) boot.esp32: SPI Mode       : DIO.[0m
.[0;32mI (81) boot.esp32: SPI Flash Size : 2MB.[0m
.[0;33mW (85) boot.esp32: PRO CPU has been reset by WDT..[0m
.[0;33mW (91) boot.esp32: WDT reset info: PRO CPU PC=0x400819f1.[0m
.[0;33mW (97) boot.esp32: WDT reset info: APP CPU PC=0x40080340.[0m
.[0;32mI (103) boot: Enabling RNG early entropy source....[0m
.[0;32mI (108) boot: Partition Table:.[0m
.[0;32mI (112) boot: ## Label            Usage          Type ST Offset   Length.[0m
.[0;32mI (119) boot:  0 nvs              WiFi data        01 02 00009000 00006000.[0m
.[0;32mI (127) boot:  1 phy_init         RF data          01 01 0000f000 00001000.[0m
.[0;32mI (134) boot:  2 factory          factory app      00 00 00010000 00100000.[0m
.[0;32mI (142) boot:  3 storage          Unknown data     01 82 00110000 000f0000.[0m
.[0;32mI (149) boot: End of partition table.[0m
.[0;32mI (154) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=28eb4h (167604) map.[0m
.[0;32mI (220) esp_image: segment 1: paddr=00038edc vaddr=3ffb0000 size=03fb0h ( 16304) load.[0m
.[0;32mI (226) esp_image: segment 2: paddr=0003ce94 vaddr=40080000 size=03184h ( 12676) load.[0m
.[0;32mI (231) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a0b14h (658196) map.[0m
.[0;32mI (459) esp_image: segment 4: paddr=000e0b3c vaddr=40083184 size=1542ch ( 87084) load.[0m
.[0;32mI (506) boot: Loaded app from partition at offset 0x10000.[0m
.[0;32mI (506) boot: Disabling RNG early entropy source....[0m
.[0;32mI (517) cpu_start: Multicore app.[0m
.[0;32mI (526) cpu_start: Pro cpu start user code.[0m
.[0;32mI (527) cpu_start: cpu freq: 160000000 Hz.[0m
.[0;32mI (527) cpu_start: Application information:.[0m
.[0;32mI (530) cpu_start: Project name:     ESPRI.[0m
.[0;32mI (534) cpu_start: App version:      v.0.2.5.[0m
.[0;32mI (539) cpu_start: Compile time:     Mar 19 2024 11:00:13.[0m
.[0;32mI (545) cpu_start: ELF file SHA256:  90da6e513....[0m
.[0;32mI (551) cpu_start: ESP-IDF:          v5.2.1.[0m
.[0;32mI (556) cpu_start: Min chip rev:     v0.0.[0m
.[0;32mI (560) cpu_start: Max chip rev:     v3.99 .[0m
.[0;32mI (565) cpu_start: Chip rev:         v1.0.[0m
.[0;32mI (570) heap_init: Initializing. RAM available for dynamic allocation:.[0m
.[0;32mI (577) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM.[0m
.[0;32mI (583) heap_init: At 3FFB8168 len 00027E98 (159 KiB): DRAM.[0m
.[0;32mI (589) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM.[0m
.[0;32mI (596) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM.[0m
.[0;32mI (602) heap_init: At 400985B0 len 00007A50 (30 KiB): IRAM.[0m
.[0;32mI (610) spi_flash: detected chip: gd.[0m
.[0;32mI (613) spi_flash: flash io: dio.[0m
.[0;33mW (616) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header..[0m
.[0;32mI (631) main_task: Started on CPU0.[0m
.[0;32mI (641) main_task: Calling app_main().[0m
.[0;32mI (641) HW/SPIFFS: Initializing SPIFFS.[0m
.[0;32mI (751) HW/SPIFFS: Partition: /storage size: total: 896321, used: 341360.[0m
ets Jun  8 2016 00:22:57
-

kamilsss655, 2024-03-19:
1. Welcome :)
2. Please fix formatting of this issue.
3. It technically should work with TTGO boards, I tried with earlier version 1.3 and it worked. Unless there are some device specific GPIO pins that cause the reset.
4. Your output log is too short, so there is no reason for reset included, so I cannot help much. Try adding the full log to the issue.

ErikS-web, 2024-03-20:
How did you flash the ESP-32 ?

I used [ESP-Launchpad](https://espressif.github.io/esp-launchpad/) (using web browser)

But beware to change the **Flash address** from 0x10000 to **0x00000** befrore pressing `program`

[image]

ErikS-web, 2024-03-20:
Here my logging from programming the version 0.2.5

and No problems....

esptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:f9:e0:e4:16:74
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Compressed 2097152 bytes to 687160...
Writing at 0x0... (2%)
Writing at 0x77c5... (4%)
Writing at 0x1d624... (7%)
Writing at 0x27ad9... (9%)
Writing at 0x2defe... (11%)
Writing at 0x342fa... (14%)
Writing at 0x3f7b8... (16%)
Writing at 0x4525e... (19%)
Writing at 0x4b11d... (21%)
Writing at 0x50c1c... (23%)
Writing at 0x56cf0... (26%)
Writing at 0x5ccc3... (28%)
Writing at 0x629b8... (30%)
Writing at 0x686e7... (33%)
Writing at 0x6da99... (35%)
Writing at 0x72e39... (38%)
Writing at 0x782ea... (40%)
Writing at 0x7d722... (42%)
Writing at 0x82dcf... (45%)
Writing at 0x885de... (47%)
Writing at 0x8e4b8... (50%)
Writing at 0x93be0... (52%)
Writing at 0x99818... (54%)
Writing at 0x9f52e... (57%)
Writing at 0xa4e98... (59%)
Writing at 0xaa8df... (61%)
Writing at 0xb0660... (64%)
Writing at 0xb756b... (66%)
Writing at 0xbce50... (69%)
Writing at 0xc28ae... (71%)
Writing at 0xc838c... (73%)
Writing at 0xd1b6b... (76%)
Writing at 0xda907... (78%)
Writing at 0xe05eb... (80%)
Writing at 0xe6164... (83%)
Writing at 0xec3cf... (85%)
Writing at 0xf1bbd... (88%)
Writing at 0x113906... (90%)
Writing at 0x1288a0... (92%)
Writing at 0x13d4f4... (95%)
Writing at 0x14c42e... (97%)
Writing at 0x15cedf... (100%)
Wrote 2097152 bytes (687160 compressed) at 0x0 in 16.196 seconds.
Leaving...

greghall-au, 2024-03-21:
Thanks for the guidance.
 I used the ESP-Launchpad, same result with the TTGP T3_V1.6, but it works nicely on a Heltec V2.
Now, where was that diagram of the connex to the Kenwood plugs?
Cheers, and I'm looking fwd to the project developing.

closed 2024-03-21 by greghall-au