kamilsss655/ESPRI #17: Add fallback AP mode

goopaga, 2024-04-10:
### Is your feature request related to a problem? Please describe.
If the device cannot connect to the wifi network (for example because of an incorrect password) after the 5 attempts there's no way to connect to the device.

### Describe the solution you'd like
After the 5 atempts start an "emergency" AP in order to reconfigure the device

### Describe alternatives you've considered
An hardware procedure to reset the configuration could also be implemented

### Additional context

kamilsss655, 2024-04-10:
Good point. I actually think the hardware reset procedure if the way to go. Just like with routers or other devices.

goopaga, 2024-04-10:
Understood.
HW reset is fine, however will you loose the entire device configuration ?

Maybe it's a little bit "hard".

In the ESPHome project (Home Assistant framework to easily integrate custom ESP nodes) the fallback hotspot is implemented for this purpose. Just as an example. 

Anyway, hw reset doors the job 👍

EA5JQP, 2024-08-15:
Hello,

I'm playing a little bit with `wifi.c` to find out if I'm able to implement a kind of emergency AP. The idea is to be able to control the ESPRI when you are temporary on the field and the configured network is not available. 

One possibility would be to have another field like `gSettings.wifi.ssid` and `gSettings.wifi.password` for AP and STA modes. A less optimal way would be to configure the Emergency AP mode via `#define` which the user cannot configure. 

What do you think?

kamilsss655, 2024-08-15:
Yup it is needed, however I'd rather not have any kind of auto fallback mode due to possible security implications (ESPRI could be remotely forced into fallback AP mode).

Rather I'd like to have a hardware reset via button - same way we have on WIFI routers.

This was main reason for the 2 hardware buttons on V2, however the BTN1 is not working on V2 due to hardware design issue, so V2 has single touch button BTN2.

I guess the solution I'd like would work like this:
* during boot user holds touch button
* LED blinks as long as button is pressed
* if the button was held for 10s factory reset is performed

This would also require touch button calibration procedure to be done only on first boot, instead of each boot (as it is done currently). Otherwise the touch button would not work, as the calibration procedure requires the touch button to be released.

This calibration change would also fix a bug in current firmware anyway - to replicate user needs to boot device while touch button is pressed and then touch button will not work until device is hard reboot.

EA5JQP, 2024-08-15:
I understand about the security implications of having an automated AP with hard coded credentials. This shouldn't be an issue if the user is able to change the password for the emergency AP right? A possibility is to ask for a password change right after the first start of the ESPRI. 

I understand the necessity of the hard reset when the user wants to go back to the default settings (when the password has been forgotten, for example).