kamilsss655/ESPRI #22: [FEATURE] Demodulate AFSK messages

kamilsss655, 2024-04-17:
### Is your feature request related to a problem? Please describe.
In order to allow messaging with these devices demodulation of signal is required.

### Describe the solution you'd like
Demodulate AFSK signal.

luisesn, 2024-08-20:
I've used succesfully LibAPRS with an esp32 to send/receive APRS, maybe its worth a look.

The only problem is that it uses audio over I2S, I've been wondering why you choose pwm over i2s.

kamilsss655, 2024-08-21:
I've seen the LibAPRS - it uses classic approach which is also what I am doing in my WIP branch.

I tried both I2S and PWM audio output and found that I2S had noticeable digital clicking/noise. It could be improved with a RC low-pass filter and tweaking the upsample rate, but not to my acceptance criteria.

PWM produced clean sound from the start.