kamilsss655/ESPRI #23: [Hardware - PCB] Orientation of  ALL 100uF Tantalum Chip Capacitors should be rotated

ErikS-web, 2024-04-17:
**Reference :** (Google search for "_tantalum elco polarisation_"..)

[image]

**Added polarisation in Schematic :**

[image]

**Resulting in :** _ALL 100uF capacitors should be rotated._

[image]

kamilsss655, 2024-04-17:
This is correct. Updating README.

Corrected in V2 board.

Well done @ErikS-web !

ErikS-web, 2024-04-17:
- Updated .CPL file into file rotated-Fixed-C1-C6-J1EXT-cpl.csv
- Wiki update with the info of this Issue #23
- New orders should be ok and info on how2  modify _already delivered boards_ is clear.

kamilsss655, 2024-04-18:
Nice @ErikS-web !

Is there any additional action to be taken in this issue? If not, then I guess it can be closed?

ErikS-web, 2024-04-18:
Yeah I will close this one ;-)

closed 2024-04-18 by ErikS-web