kamilsss655/ESPRI #25: [FEATURE] A TX-start-delay and a tiny Tail-delay might be usefull

ErikS-web, 2024-04-23:
Tested with V8.0

[image]

[WAV beacon in Release v0.8.0.zip](https://github.com/kamilsss655/ESPRI/files/15077275/WAV.beacon.in.Release.v0.8.0.zip)

closed 2024-04-24 by kamilsss655

ErikS-web, 2024-04-24:
Yes this is OK

[image]

[ok.zip](https://github.com/kamilsss655/ESPRI/files/15099074/ok.zip)

kamilsss655, 2024-04-24:
Cool! Here is spectrogram recorded vs original:
[image]

Although it is likely distorted due to mp3 compression of the recording.

commit bee6f245 (2024-04-24): fixes #25 ptt on/off delay of 500ms