kamilsss655/ESPRI #28: [BUG] MORSE timing is not correct on SPACES

ErikS-web, 2024-04-24:
Spaces between dits and dahs within a letter code are the length of one dit, spaces between letters in a word are the length of three dits (one dah), and spaces between words are seven dits long.

One line = 3 points.
The space between the dots and dashes of one and the same Morse character = 1 point.
The space between two Morse characters = 3 points.
The space between two words or numbers = 7 points.

reference
cannot find the English version of https://nl.wikipedia.org/wiki/Morse

kamilsss655, 2024-04-25:
This is correct.

The morse code beacon is not fully featured, as it doesn't even decode letters and numbers to di dahs. So it doesn't have a concept of words. The input is series of "." "-" " ".

So this can for sure be improved, however is not the highest priority right now.

In the meantime you can add spaces in between words to achieve the desired effect. For example:
`- . ... -     .--- ..- ... -    - . ... -`

ErikS-web, 2024-05-06:
A nice site explaining morse timing.
https://www.mobilefish.com/services/morse_code/morse_code.php

When I increase baudrate, timing is very bad and Espri seems to drop the "dots"

kamilsss655, 2024-05-23:
Yup I've noticed that as well, the high baud Morse code might require refactor from `ms` to `us` as timing units, the same way as AFSK modulation was done.