kamilsss655/ESPRI #29: [BUG] C7 capacity is too big

kamilsss655, 2024-04-25:
### Describe the bug
The audio in voltage takes 12-16s to drop off after the audio is switched off. This is because C7 capacitor is too big (gets discharged through the R7 (100k).

https://www.digikey.com/en/resources/conversion-calculators/conversion-calculator-time-constant

### Expected behavior
Audio voltage drops instantly.

### Details
ESPRI V1 pcb.

### Fix
Ideally the C7 capacitor should have 100nF.

To fix this you can replace C7 capacitor with 2x220nF capacitors of 1205 size connected in series. Of course there are multiple ways to approach this. However this is what I have personally done on my boards.

@ErikS-web can you look into replacing C7 in CPL and BOM files into C145085 ? It is 1uF capacitor with same footprint, so that would make more sense I think.

kamilsss655, 2024-04-25:
Also for BOM and CPL files I think we should change voltage bias resistors from 100k to 10k to make RC time constant lower, compensating for the too big capacitor.

This will increase current consumption in deep sleep, but will make the boards functional.

kamilsss655, 2024-04-25:
Fix:
[image]

ErikS-web, 2024-04-26:
Placed this info on Wiki ([Value of C7 should be 100nF](https://github.com/kamilsss655/ESPRI/wiki/30-%E2%80%90-Hardware-and-connections#value-of-c7-should-be-100nf-made-by-2x-220nf).

With 'Note'; Modification is not yet present in our BOM-files. So you have to do it manually.

kamilsss655, 2024-04-27:
Thanks Erik.

I don't think this will be a huge issue, as this only affects how long it takes to turn off the squelch. It should not affect demodulation etc. as essentially this capacitor along with the high input impedance of the audio in circuit creates a high pass filter. In this case the pass band is too wide for the use case. Thus replacing the capacitor with 1uF for thenew boards should be sufficient.

New release v0.9.0 has audio input detection / squelch so users can test it themselves.

kamilsss655, 2024-04-28:
BTW fixed schematic.pdf alond with BOM and CPL files.
C7 in schematic is 100nF, and it is 1uF in BOM and CPL files. Both should be fine.

ErikS-web, 2024-05-23:
Wiki is updated also. Suggest to close this Issue

closed 2024-05-23 by kamilsss655