kamilsss655/ESPRI #38: [FEATURE] Future Library

georgio, 2024-05-06:
It will be super interesting if your module can communicate with hosts on all sorts of platforms.

Perhaps we can create a low-level library using something universal like Rust. This library could then be imported by both web (through wasm) or iOS/Android apps, or whatever other platform.

The end result could be that we can control the UV-K5* through a mobile app or even remotely, and then the problem of maintaining the ESP32 interface could be separated from the development of such apps.

I'll be very happy to contribute to building this with you.

kamilsss655, 2024-05-06:
Hi @georgio, thank you for you interest in this project. I am open to help from other devs with this project.

If I understand correctly you'd like to provide some form of universal interface to the ESPRI platform. It could be feasible, but I think there already is such interface in the form of the REST API. Adding another layer (Rust) on top of already abstract interface seems like necessary overhead, unless I misunderstood your concept.

There are bunch of things to be worked on like UART comms, modulation and demodulation, frontend work, PCB design of V2 and much more. I'd take PRs for any of those if you'd be interested in that.

So for there has been 0 PRs submitted for this project, so if you'd like this project to grow please spread a word about. Hopefully this brings new talent to the project. In the meantime I will be putting in on pause until the project has better recognition and engagement amongst the communities.

closed 2024-05-06 by kamilsss655

georgio, 2024-05-09:
Yeah that's very reasonable. Thanks for the response.
I will try to help out (and will probably bother you with some questions).

Once I have some time, I will try to construct the v1 PCB myself.