kamilsss655/ESPRI #39: [FEATURE] Through hole version for ESP32 DOIT board with OLED and SWs

r4d10n, 2024-05-10:
We have been playing with Quansheng UVK5 and was looking for something with ESP32 for AIOC like functions. Now this is a very interesting project with a lot of potential usecases !

Since Lolin32 Lite is not common in this part of the world and SMD sourcing / assembly is still cumbersome, we've attempted a Through Hole version for the most common 30 pin ESP32 DOIT board.

Changes:

- Since there are a lot of unpopulated I/Os, we've added an 0.91" OLED, couple of LEDs, Switches and one more Touchpad Button. 
- The ADC/DAC IO pins have been broken out to a header
- Header Pins for Radio PWR, 5v and 3.3v 
- SMD 2N7002 has been replaced with THT 2N7000  

Kindly do check and review the schematics and PCB documents so that we can attempt a build of the board pretty soon. 

Repo: https://github.com/r4d10n/esprit

[[Interactive BOM]](https://htmlpreview.github.io/?https://github.com/r4d10n/esprit/blob/main/ibom/ibom.html) - [[Schematics]](https://kicanvas.org/?github=https://github.com/r4d10n/esprit/blob/main/espri-doit.kicad_sch) - [[PCB]](https://kicanvas.org/?github=https://github.com/r4d10n/esprit/blob/main/espri-doit.kicad_pcb) - [[Gerbers]](https://github.com/r4d10n/esprit/blob/main/production/espri-doit.zip)

Renders:

[image]

[image]

[image]

[image]

kamilsss655, 2024-05-10:
Hey!

That's really cool! The schematic looks good to me. :+1: 

Some of the suggestions/notes:
* it seems that there is ground-plane covering the area of the ESP antenna - this might affect WIFI performance
* the LDO of the ESP board used has relatively high quiescent current, so the battery life during deep sleep will be affected
* I experimented with adding uSD card slot for the V2, however swapping FLASH to 16MB might be more than needed to keep multiple audio samples for example
* you might consider adding 3.3k and 100nF LPF in series to the audio output to further improve audio output and eliminate audio glitching during times of silence

Overall looks really good and happy to see another version of ESPRI board. Goodluck!

r4d10n, 2024-05-12:
Thank you for the detailed review ! 

- Initially the LED and TPs were kept away on the board side with empty ground plane below the antenna, which affected the compactness. Will look into rerouting the same in the next revision.
- Since OLED is also included and WiFi performance seems to be affected with low Iq LDOs, low power battery operation may not be ideal with this board.  Are you getting issues with HT7333 for WiFi performance ?
- SD card will be a nice addition. WROOM modules have Flash inside the the shield :(  We could also look into [Codec2](https://github.com/deulis/ESP32_Codec2) or [Speex](https://github.com/UncleRus/esp-idf-speex/) for compressed storage.
- Will add in the HPF..

Looking forward to experimenting ! 

Cheers

kamilsss655, 2024-05-23:
Closing this one as it is not exactly an issue. Keep me posted though on any advancements etc.

closed 2024-05-23 by kamilsss655