kamilsss655/ESPRI #43: can not set local wifi with the esp32

chinesejacklee, 2024-05-20:
Hi there, my first comment, great work! how do I set the esp32 so it can be connected with my local WIFI pls? I can't find anything in the user manual talks about it. I tried to change the wifi to the 'station mode', but I am now stuck and can not connect with the the http://192.168.4.1/ anymore. Can anyone help pls?

ErikS-web, 2024-05-20:
Did you had a look at the logging... via `Console` that is available in  [`🗲ESP-launchpad🗲`](https://espressif.github.io/esp-launchpad/)

There you can see under what `(new) ip-adress` the ESP is connected in `your home-network`
Look for GREEN LINE with **esp_netif_handlers: sta ip:** 

You can also have a look in your router-modem.

p.s. You can always reprogram the ESP, and try again

kamilsss655, 2024-05-22:
Thanks Erik!

closed 2024-05-22 by kamilsss655