kamilsss655/ESPRI #45: [BUG] Parts cannot match the corresponding pads on the PCB board (JLCPCB assemble)

jenccnej, 2024-06-03:
### Describe the bug
I placed a JLCPCB order with assemble using the latest version of the board
https://github.com/kamilsss655/ESPRI/tree/main/hardware/v2

They sent this kind of feedback:

"Dear customer,
Well got your order with many thanks.

Sorry to bother you, but there is one thing that we want to confirm with you about your PCBA order SMT02406011770241_Y1 before proceeding.
As shown below, seems the packages of the parts (

R3,R4,R5,R6,R7,R8,R11,R14
) cannot match the corresponding pads on the PCB board, so we are unable to assemble them."

[image]

### To Reproduce
Steps to reproduce the behavior:
Place a JLCPCB order with assemble using
1. Following instructions
https://github.com/kamilsss655/ESPRI/wiki/30-%E2%80%90-Hardware-and-connections#how2-order-pcb-v1
2. Following files
https://github.com/kamilsss655/ESPRI/tree/main/hardware/v2

### Expected behavior
JLCPCB order accepted into production

kamilsss655, 2024-06-03:
JLCPCB is correct. I have made a mistake of selecting 1206 instead of 0805 size resistor (should be `C149504`).

This issue in the `bom.csv` is now corrected in the main branch.

Thanks for feedback.

kamilsss655, 2024-06-03:
It is interesting though since I have ordered mine (which is already on it's way) with the same issue, but they didn't reach out to me. Wonder if they replaced the resistors with the correct size on their own. Will know once I receive my boards :)

closed 2024-06-03 by kamilsss655