kamilsss655/ESPRI #46: Message sent from one radio to another can not be displayed/seen anywhere in the webportal or the serial port

chinesejacklee, 2024-06-07:
1. connected ESPRI board with the radio B
2. sent a msg from radio A to radio B. I can confirm the msg has been received from the radio B screen
3. there is no way for me to see the received msg anywhere in the esp32 serial port or the web portal
4. I am not using ESP32 Lolin Lite, instead I am using an ESP32 Wrover,  but by following the same pins connections in the WIKI
5.  not sure if it will help to locate the root reason, when I tried :
curl -d '{"content":"Hello!"}' http://192.168.4.1/api/uvk5_message
it shows me:
{
        "response":     "Invalid JSON received"
}
[image]
[image]
[image]
[image]

chinesejacklee, 2024-06-07:
https://github.com/kamilsss655/ESPRI/assets/2108092/8d421f27-74af-4b8a-bd59-c32eeafe31ac

here is a video that tries to demonstrate the issue I have

kamilsss655, 2024-06-07:
Yea the K5 messaging is a bit of a hidden/unfinished feature. At the moment it only supports my K5 firmware version [v.21.4](https://www.patreon.com/posts/release-v-21-4-102196542) .

This has to do with the ESPRI serial baud rate which is currently hardcoded `115200` to match the `v.21.4` of the K5 firmware.

You might be able to make it work with other K5 firmwares by compiling custom version of ESPRI with a different baud rate.

kamilsss655, 2024-06-07:
In the future this would be resolved by the https://github.com/kamilsss655/ESPRI/issues/2 where serial baud rate would be configurable from the web client.

closed 2024-06-07 by kamilsss655

chinesejacklee, 2024-06-08:
@kamilsss655 thanks, just to confirm that the ESPRI currently does not support 'forward' received message to the esp32 board, that in turn can pass the msg to a PC? like what you draw on the diagram below? Ultimately, I am trying to use a radio to turn on/off things via esp32/pc. Any suggestion if it is not supported currently, or is it on the future road map pls? Thanks
[image]

chinesejacklee, 2024-06-08:
@kamilsss655 if I am not wrong, I think somehow I have an impression that the ESPRI enbles ESP32 to display incoming msg to the web pannel. Is this correct, happy to become a patreon and get the firmware 21.4 if what I am after is supported. Can you confirm pls?

kamilsss655, 2024-06-08:
The received message currently shows up in the web panel in the notifications page and in serial monitor output of the ESP, so it is displayed, but it is more like a debug form and not a commercial user experience.

chinesejacklee, 2024-06-08:
@kamilsss655 so you think v21.4 will provide a better user experience in term of presenting the msg?

kamilsss655, 2024-06-08:
No, the v21.4 will allow you to receive a message from K5 on ESPRI.

You can also try this without v21.4 by connecting USB serial dongle to ESPRI (RX-TX, TX-RX, GND) and sending text from a PC via serial dongle to ESPRI (use 115200 baud).

chinesejacklee, 2024-06-10:
ok, just become one of your patreon! I will give the 21.4 a go

chinesejacklee, 2024-06-11:
@kamilsss655 installed 21.4, I can see the message in the serial monitor now! but the API still does not work, see screenshot below pls:
[image]

chinesejacklee, 2024-06-11:
@kamilsss655 was the way I invoked API correct?

kamilsss655, 2024-06-11:
Interesting, might be that your `curl` on Windows is sending something else. I was able to send message with:
```
curl -d '{"content":"hello this is just a test ok ok ok"}' http://ESP_IP/api/uvk5_message
```
on Linux.

Maybe use some other client i.e. [Postman](https://www.postman.com/) to send request.

chinesejacklee, 2024-06-12:
@kamilsss655 I was able to do that in the linux, postman in windows does not work by the way. I thought the curl -d '{"content":"hello this is just a test ok ok ok"}' http://ESP_IP/api/uvk5_message will send a msg to another radio, but apparently apart being able to see it in the serial port, the msg does not go out to another radio, am I right? what is the use cases for this API feature pls? thanks! good work by the way!

kamilsss655, 2024-06-12:
> I thought the curl -d '{"content":"hello this is just a test ok ok ok"}' http://ESP_IP/api/uvk5_message will send a msg to another radio

It does send the message (you should see K5 red led as PTT indication), but sometimes it might require you to send the request couple of times. This is known issue with K5 serial handling which sometimes drops incoming serial data.

chinesejacklee, 2024-06-13:
tried more than a few times, but the msg was not able to send the msg to another radio 
[image]

chinesejacklee, 2024-06-13:
@kamilsss655 I am sending the msg from curl, as you can see, the msg was presented correctly in the serial port, but the msg was not sent out. Anything else I can try pls?

kamilsss655, 2024-06-14:
Looks like K5 is not receiving the serial command. Not sure why. You might want to try using USB serial dongle (one used for programming K5) and sending:
```
SMS: message content\n
```