kamilsss655/ESPRI #50: [FEATURE] V2.2 board battery measurement

kamilsss655, 2024-06-09:
### Is your feature request related to a problem? Please describe.
New 2.2 boards support battery measurement.

### Describe the solution you'd like
Battery voltage is measured every couple of minutes and the reading is available in `About` page in frontend app.
Voltage divider is only powered during the measurement and then turned off via MOSFET to conserve power.

KDMcMullan, 2024-07-06:
From the schematic, it looks like it's already added.
HOWEVER, your divider is 200k total. That's drawing like 8.2 / 2e5 = 41uA.
That portion of the circuit should not deplete a 2200mAh battery for at least 2.2 / 4.1e-5 = c. 6 years.
I could be completely mad, but I'm not sure it's worth the effort.

kamilsss655, 2024-08-03:
There is no need to drain battery all the time, if the voltage measurement can be taken in less than a second every few minutes.

Moreover you should note that there are other elements already on the board that deplete the battery:
1. The LDO quiescent current
2. Audio input voltage divider

The board is designed to be battery efficient. This is also the reason this particular ESP host board has been selected, as it offers good price to LDO quiescent current ratio. And in my book the more efficient the better.

KDMcMullan, 2024-08-03:
I completely agree. There's no need.But with the calculated drain being so minimal, it's not worth the extra effort of adding even a 10c transistor.