kamilsss655/ESPRI #58: [FEATURE] [DISCUSSION] MQTT Implementation

KDMcMullan, 2024-07-06:
As a predominantly IoT project, I'd like to see an MQTT implementation.

I have a strong dislike for IFTTT, in fact I religiously avoid cloud-based IoT solutions. I (almost exclusively) run [TASMOTA](https://github.com/arendst/Tasmota) and I'm extremely pleased with it. The only place my IoT network touches the internet is the [Home Assistant](https://www.home-assistant.io/) web interface.

This thread is suggested as a discussion document as to what an MQTT implementation might look like. One suggestion is that the firmware might subscribe to an MQTT topic on a given broker,  which publishes a list of "commands" and "queries". This would allow those commands and queries to be dynamically published on the web server or by MQTT direct.

For example, an IoT controller might publish
```
espri/dict [
{"cmd":"true","type":"bool","title":"Bathroom Lights","pub":"bathlights/power","payloads":"['on','off']","sub":"bathlights/stat/power"},
{"cmd":"false","type":"float","title":"Garden Temperature","pub":"garden/status","sub":"garden/stat/temp"}
]
```
This creates a boolean button, allowing on/off of lights, and a numeric display of temperature, along with the MQTT publications / subscriptions required to populate the display, and take commands from the web server.

kamilsss655, 2024-07-06:
Sounds good to me. I'd consider taking PR for such a feature.

KDMcMullan, 2024-07-06:
I've used ESP32 here to pub / sub MQTT. C is not my best language. Is the ESPRI PCB available to buy anywhere? I'd have a stab at it.