kamilsss655/ESPRI #69: espri case

ascarons22, 2024-08-18:
oh well, design a carcasa for the spirit, I say gonzalo that you envy for that, keep the archives stl inside the carpet that adjunto.un saluto
[Archivos.zip](https://github.com/user-attachments/files/16651943/Archivos.zip)

kamilsss655, 2024-08-19:
That's cool!

Do you have any pictures of a printed case by any chance?

Can you specify under which license you're providing this case work? If you haven't thought about it yet I'd suggest 'Apache-2.0 license'.

ascarons22, 2024-08-19:
[image]
[image]
[image]

kamilsss655, 2024-08-22:
That's really cool! Thanks for sharing. :rocket:

ErikS-web, 2024-10-19:
Very nice. 

Added Case to [the Wiki](https://github.com/kamilsss655/ESPRI/wiki/80-%E2%80%90-Tips-&-Tricks-and-Known-Issues#espri-case)